makerbase-mks/MKS-Robin-E3P #1: It is not printing. Don't heat the bed

carlosrondini, 2020-10-22:
I have an ender 3 with MKS-Robin-E3P and TMC2209.
I followed the settings:
Set default_envs = mks_robin_e3p
Set #define SERIAL_PORT -1
set #define SERIAL_PORT_2 3
Set #define MOTHERBOARD BOARD_MKS_ROBIN_E3P
#define INVERT_X_DIR false
#define EEPROM_SETTINGS
#define X_DRIVER_TYPE TMC2209

When used MKS UI does not print, does not heat the bed.
But, using 128x64 Marlin UI or Color Marlin UI works.

I used:
Mks-Robin-Nano-Marlin2.0-Firmware
marlin2.0.7

mks-viva, 2020-10-23:
Can be check temp on MKS TS35 screen when you use MKS UI?

carlosrondini, 2020-10-23:
> Can be check temp on MKS TS35 screen when you use MKS UI? Can you provide some picture when you use MKS UI printing.

Yes. It stays at room temperature, does not increase.

mks-viva, 2020-10-27:
Please provide your use the gcode file to me.

carlosrondini, 2020-10-27:
[CE3_CableClip1.zip](https://github.com/makerbase-mks/MKS-Robin-E3P/files/5442430/CE3_CableClip1.zip)

mks-viva, 2020-10-27:
Ok! We will test it.

mks-viva, 2020-10-28:
MKS Test it is OK. Can you format your TF card?

carlosrondini, 2020-10-29:
Hello, unbelievably, it worked.
Thank you.

mks-viva, 2020-10-29:
OK! We will close it.

closed 2020-10-29 by mks-viva