makerbase-mks/MKS-Robin-E3P #4: MKS PWC

Surax98, 2020-11-17:
I can't properly set up my PWC module.I am following the guide you posted, but I'm unable to make it works. I tried to remap my pins too, but nothing seems to work. Is it a bug or am I just missing something?

mks-viva, 2020-11-20:
Can you provide your motherboard wirings to me?

Surax98, 2020-11-20:
[image]
This is how i connected signal pins. I can assure the rest is correctly connected (220V plug to input and 220V output to power supply paying attention to not swap L and N. Logic power to 5V/GND on the same connector as the signal pins). When I unplug the printer and then replug, the outage detection seems to work and in fact, it resumes from where stopped, but I can't make it work with the M81 code. Just says "*3d printer name* OFF" and keep being ON. I attach how i defined pins too.

[image]

mks-viva, 2020-11-21:
MKS PWC's PW_DET connect to MKS Robin E3P's MT-DET
MKS PWC's PW_OFF connect to MKS Robin E3P's PB0
So, you pin define like this:
#define KILL_PIN    PA4
#define PS_ON_PIN   PB0
[image]

Surax98, 2020-11-21:
I tried to follow defines from your screenshot, but it didn't work. I'll try to follow the ones you posted and let you know

mks-viva, 2020-11-21:
Can you provide your MKS Robin E3P's config to me and i config it and it work ok, i will send it for you.

Surax98, 2020-11-21:
[Marlin E3P.zip](https://github.com/makerbase-mks/MKS-Robin-E3P/files/5577327/Marlin.E3P.zip)
I attached everything, so you can check every file (configuration, adv_config, pins, ecc ecc). Just let me know the quickest possibile, including how to physically connect pins correctly

Surax98, 2020-11-21:
Meanwhile, I tried to define pins in this way:
#define KILL_PIN PA4
#define PS_ON_PIN PB0

on PA4 I have PW_DET, on PB0 I have PW_OFF. still can't shut down printer using M81

Surax98, 2020-11-23:
@mks-viva Any news? Please, I need to make it work...

mks-viva, 2020-11-23:
Hi! I has test it
1. because you has use Marlin color UI and you has enable #define TFT_480x320_SPI, so you need settings pin in pins_MKS_ROBIN_E3P.h file;
[image]

Surax98, 2020-11-23:
@mks-viva can you please provide me the right pin file so i have no chance to do something wrong?

mks-viva, 2020-11-23:
I upload pins file to here, you can download it.

Surax98, 2020-11-23:
@mks-viva I used the pin file provided but sending M81 still keeps my boards ON.... LCD keeps being on, extruder fan keeps being on, everything keeps being on... Just extruder and heated bed stops being hot (but i think this is just because temps are set to 0 through software command.....

Surax98, 2020-11-23:
I fixed it! I digged into M80-M81 gcode file (M80_M81.cpp) and found out this thing:
[image]
In my pins file, I had SUICIDE_PIN set to PB2, which, apparently, doesn't exist on the E3P (or i can't find it on the pinout diagram)
[image]
So, i just commented out this pin, forcing M81 to use the function PSU_OFF() instead of suicide() and that worked!

mks-viva, 2020-11-24:
@Surax98 
So, this problem are solved?

Surax98, 2020-11-24:
Yes, solved

Surax98, 2020-11-24:
Yes, solved

closed 2020-11-24 by Surax98