makerbase-mks/MKS-Robin-E3P #6: Bltouch issue

edouardkleinhans, 2020-12-28:
Hello, 

I can't add a bltouch to this motherboard.

bltouch is connected to PA8 and endstop to Z endstop. 

Can you help me on this ?

[Configuration_adv.h.txt](https://github.com/makerbase-mks/MKS-Robin-E3P/files/5747428/Configuration_adv.h.txt)
[Configuration.h.txt](https://github.com/makerbase-mks/MKS-Robin-E3P/files/5747429/Configuration.h.txt)

mks-viva, 2020-12-29:
Hi! You can refer to here: https://github.com/makerbase-mks/MKS-Robin-E3P/tree/master/Firmware
[image]

edouardkleinhans, 2020-12-29:
That's what I try, and it doesn't work.

mks-viva, 2020-12-30:
Can you provide your wiring picture to me?

edouardkleinhans, 2020-12-30:
I've use latest marlin fw (2.0.7.2) and i work now. So I think you have an issue on your build.

mks-viva, 2020-12-30:
Thank you for your feedback, but we have tested the firmware and it works normally.

closed 2021-01-02 by edouardkleinhans