makerbase-mks/MKS-Robin-E3P #8: MKS UI -  Random movement of the selected option in the GUI.

myckiecat, 2021-01-07:
Hi,

Recently i purchased a MKS Robin E3P board with TFT and WIFI module.
All was working fine, but MKS UI have started to doing random things selecting objects from the UI.
It seems to happens when any motor starts to move.
In a clean boot, menu is showed correctly, but after i do home, or move some motor, highlited option in the UI gets randomly selected (but not executed).
I have installed two Z motors to the motherboard.

I think it's related to some noise coming from drivers or hardware error on the touch screen, or perhaps a 
software bug, i dont know, and want to know what do you think about it.

Video is attached:

https://user-images.githubusercontent.com/77124821/103942100-266eae80-5130-11eb-95b0-62bd7f6faca0.mp4

Thanks.

mks-viva, 2021-01-08:
Hi! I want to know if you only use the touch function, if it is normal.

myckiecat, 2021-01-08:
> Hi! I want to know if you only use the touch function, if it is normal.

Hi,
I use stock firmware that mks give to this board (TMC2209 + UART) and with it i can use touch and the button, but usually only use touch function.
More testing done, and i think problem is the second motor of Z axis, if i disconnect second z axis motor, this random movement in the items stops at all.
Perhaps board or driver can not handle this two motors? Anyway, i think we can close issue, not software related at all.

Thanks.

mks-viva, 2021-01-09:
What is your current motherboard Dirver IC power setting?
[image]

myckiecat, 2021-01-09:
> What is your current motherboard Dirver IC power setting?

Current settings are:

X/Y - 550 mA
Z - 800 mA
E - 650 mA

Stock settings that MKS give with the firmware is too high (800 mA for all), all motors get very hot with default values you provide in Ender 3 stock motors.

Thanks.

myckiecat, 2021-01-16:
> > What is your current motherboard Dirver IC power setting?
> 
> Current settings are:
> 
> X/Y - 550 mA
> Z - 800 mA
> E - 650 mA
> 
> Stock settings that MKS give with the firmware is too high (800 mA for all), all motors get very hot with default values you provide in Ender 3 stock motors.
> 
> Thanks.

Any comment or solution to problem???

Thanks.

myckiecat, 2021-01-24:
> Any comment or solution to problem???
>
> Thanks.

After changing some components, cooling fan for mks board, to a Noctua 4010, the problem was gone...
I'm closing this issue because i can not reproduce it anymore.

Thanks.

closed 2021-01-24 by myckiecat