makerbase-mks/MKS-Robin-E3P #9: Lot of options missing in E3P TFT UI for ender 3

myckiecat, 2021-01-16:
Hi,

When i see your videos of TFT35 i see a lot more options and info in UI with TF35 that with the TFT included in E3P package.
Can you update UI to have at least more info, like position of nozzle, enter GCODE commands menu, more PREHEAT filament options (only 2 is not enought for me).

Today, i'm trying to install your 3dtouch with Robin E3P, and i can not find how to setup it correctly, i think you missing information for this UI of the ROBIN E3P to config 3dtouch offsets.

Thanks.

mks-viva, 2021-01-18:
Hi! Some parameters cannot be modified directly and have an effect. They need to be modified and compiled in the firmware to take effect.

myckiecat, 2021-01-18:
> Hi! Some parameters cannot be modified directly and have an effect. They need to be modified and compiled in the firmware to take effect.

But, for exemple, Why dont have Z value in MOVE menu? This have no sense. Other UI's have posibility of see Axis values when you are moving it.
All Value axis are important in MOVE menu, because then you can ajust and test each axis movement and steps.

mks-viva, 2021-01-19:
This function is to add axis information display.

closed 2021-01-25 by myckiecat