makerbase-mks/MKS-Robin-E3P #10: Not enought precision in Z offsets value in MKS UI.

myckiecat, 2021-01-17:
I think there is a problem with accuracy in Z offsets when setting up a MKS Touch.
You put online one decimal digit precision in this value in the UI and this is not enought to calibrate offset with precision...
Please update, because almost two digits are necesary to setting up a true zoffset in your MKS UI.

If not, i'm not able to setup your MKS Touch with this board properly.
I tried modifying this value in marlin , but i dont know if it's applied correctly the offsets, because when i go to MKS UI Z offsets menu, i see a rounded value....

Thanks.

MKS-Sean, 2021-01-18:
I got this bug and I will fix it, because we usually use babystep to adjust the value of zoffset, which is rarely used in parameter settings. You found this problem

MKS-Sean, 2021-01-18:
@ Fixed this problem

myckiecat, 2021-01-18:
> I got this bug and I will fix it, because we usually use babystep to adjust the value of zoffset, which is rarely used in parameter settings. You found this problem

Thanks!!

myckiecat, 2021-01-18:
> @ Fixed this problem

Tested it and now it's really nice :)
Thank you for fixing it.

closed 2021-01-18 by myckiecat