makerbase-mks/MKS-Robin-E3P #22: No compila MKS-Robin-E3P-master / Source-code

BenjaGaldames, 2021-08-28:
Hola,

Tras descargar el archivo y generarlo con VSCode / PlatformIO / Marlin Firmware, me da este error.

Ayuda por favor !!!

[image]

mks-viva, 2021-08-31:
Please provide your source code to me and i need test it.

BenjaGaldames, 2021-08-31:
Hola,

Código fuente en el enlace.

https://drive.google.com/file/d/19gOYfWlFfZKFq_7VERm1705jKhZPbpVN/view?usp=sharing

[image]

mks-viva, 2021-09-01:
I has compile it and it's ok. Maybe you need to try a few more times.
[image]
[Robin_e3p.zip](https://github.com/makerbase-mks/MKS-Robin-E3P/files/7088042/Robin_e3p.zip)

BenjaGaldames, 2021-09-02:
Hola,

Al compilar en otra computadora, me funciono, al parecer no funciona en todas las computadoras.
El problema que tengo ahora es que los finales de carrera no me los toma en cuenta, tengo una CR-10 en modo Dir.

Ayúdame con los finales de carrera.

https://drive.google.com/file/d/1EKTz9yING4X7zVFF-XFmzvSjxvPUVdJJ/view?usp=sharing

mks-viva, 2021-09-02:
[Robin_e3p.zip](https://github.com/makerbase-mks/MKS-Robin-E3P/files/7096618/Robin_e3p.zip)

BenjaGaldames, 2021-09-07:
Aun no me funcionan los finales de carrera.

Necesito una solución para que funcionen los finales de carrera.

closed 2021-09-10 by BenjaGaldames